mathertel/OneButton #5: did not work on attiny85

kellerkind, 2015-09-04:
SimpleButton Example not work on attiny85 (Using arduino core frome [here](http://highlowtech.org/?p=1695)). Changed the output pin to 0 (D0) and the kept the input to A1 (D2). Neither Double Click nor Single Click works for me. Is the milli() function supported on attiny?

Changed code:

	/*
	 This is a sample sketch to show how to use the OneButtonLibrary
	 to detect double-click events on a button. 
	 The library internals are explained at
	 http://www.mathertel.de/Arduino/OneButtonLibrary.aspx
	  
	 Setup a test circuit:
	 * Connect a pushbutton to pin A1 (ButtonPin) and ground.
	 * The pin 13 (StatusPin) is used for output attach a led and resistor to ground
	   or see the built-in led on the standard arduino board.
	   
	 The Sketch shows how to setup the library and bind a special function to the click event.
	 In the loop function the button.tick function has to be called as often as you like.
	*/
	 
	// 03.03.2011 created by Matthias Hertel
	// 01.12.2011 extension changed to work with the Arduino 1.0 environment

	#include "OneButton.h"

	// Setup a new OneButton on pin A1.  
	OneButton button(A2, false);

	// setup code here, to run once:
	void setup() {
	  // enable the standard led on pin 13.
	  pinMode(0, OUTPUT);      // sets the digital pin as output
	  
	  // link the click function to be called on a click event.   
	  button.attachClick(click);
	} // setup
	  

	// main code here, to run repeatedly: 
	void loop() {
	  // keep watching the push button:
	  button.tick();

	  // You can implement other code in here or just wait a while 
	  delay(10);
	} // loop

	// this function will be called when the button was pressed 2 times in a short timeframe.
	void click() {
	  static int m = LOW;
	  // reverse the LED 
	  m = !m;
	  digitalWrite(0, m);
	} // click

	// End

kellerkind, 2015-09-04:
Update: 

The example works when compiling for attiny85@1Mhz. It does not work when compiling attiny85@8Mhz.

UnexpectedMaker, 2018-07-08:
Just ran into this... wasted a good 2 hours thinking it was me :(
Will this ever be fixed? Issue is Super old..
Thanks!

mathertel, 2021-01-24:
please check again with version 2.0.0. should be fixed.

closed 2021-01-24 by mathertel