mathertel/OneButton #6: Where did `attachLongPressStart` go?

cottsak, 2015-09-16:
I'm interested in using `attachLongPressStart` as per https://github.com/mathertel/OneButton/blob/master/examples/TwoButtons/TwoButtons.ino however `OneButton.h` only defines:

```
  void attachClick(callbackFunction newFunction);
  void attachDoubleClick(callbackFunction newFunction);
  void attachPress(callbackFunction newFunction);
```

cottsak, 2015-09-16:
The problem appears to be that the `OneButtonLibrary.zip` hosted on http://www.mathertel.de/Arduino/OneButtonLibrary.aspx is the wrong version. Please update the link to use the "Download ZIP" from Github: https://github.com/mathertel/OneButton/archive/master.zip

closed 2015-09-16 by cottsak

mathertel, 2015-09-21:
fixed. I directly point to the download file on GitHub.
I will add the library to the Library Manager soon....