mathertel/OneButton #7: Problems with long press on esp8266

cottsak, 2015-09-16:
I have a simple push button pulling LOW on GPIO2 as per this code:

```
#include <OneButton.h>

OneButton gpio2Input(2, true);
  
void setup() {
  // attach handler under assumption relay will hold on
  gpio2Input.attachLongPressStart(alarmIsSounding);
  
  Serial.begin(115200);   // for debugging
  Serial.println("\r\n*** Ready to rock! ***");
}

void loop() {
  gpio2Input.tick();

  // ...
}

void alarmIsSounding(){
  Serial.println("alarm!");
}
```

The problem is that I'm only occasionally getting "alarm!" in my serial monitor. I'm long-pressing for various durations (all > 1000ms) and only some of the messages get into the monitor. From reading https://github.com/mathertel/OneButton/blob/master/OneButton.cpp line 128 I'm assuming that every long press (>1000ms) I should see the logging after the 1000ms mark. Is that right?

Maybe it's the esp8266 I don't know. Buy one on ebay: they're like $4 and you can test it yourself. Thanks

mathertel, 2015-09-21:
That really sounds strange and I never have had effects like that on my boards.

cottsak, 2015-09-22:
Still having the problem. Any chance you can investigate/repo it?

cottsak, 2015-09-22:
I haven't tested it for very long but it looks like this change https://github.com/mathertel/OneButton/commit/4ef85ac052c0307615c1ea71b55e1ce3bf31f189 may have helped the problem in this issue.

closed 2017-05-14 by mathertel