mathertel/OneButton #21: Click detected at doubleclick

xydix, 2017-05-17:
Hi.
Just found this. It is awesome to get three functions from one button.
I use this with mysensors library to create a smart button.
I got one problem..
Sometimes the i press the button one time it is detected as a doubleclick..
I tried with two switches, both of the same type but i still get this. Is there any values to modify to solve this?

xydix, 2017-06-26:
After update to 1.2.0 it seems better. I will test for a while and let you know.....

S4mw1s3, 2017-09-12:
I've also been testing 1.2.0 now for a while and I haven't encountered a situation yet where a click/double click was wrong. Is it possible to close this issue or update it as needed? Thanks.

xydix, 2017-09-12:
Sorry. Forgot about this.
I would say it is rock solid after updating to 1.2.0.

closed 2017-09-12 by xydix