mathertel/OneButton #22: Map same callback to multiple buttons and handle button presses using button id parameter passed into call back.

danobot, 2017-05-19:
I'm building an arduino with 20 buttons. Each callback is handled exactly the same, the only difference being the message displayed on screen.
I was wondering if it is possible to assign the same call back to multiple buttons and then differentiate between different buttons inside the callback using a button id.

idolpx, 2017-07-25:
I like this idea.  :)   I may add it myself if you haven't already.

idolpx, 2017-07-25:
The pin number could be the button id.

danobot, 2017-07-26:
ive already implemented this functionality. cant remember how much moditifcation of the base class was required. can push a pull request and @mathertel can review

danobot, 2017-11-30:
sorry, will create PR soon.

closed 2017-11-30 by danobot