mathertel/OneButton #24: Make this work with port expanders?

kvoit, 2017-09-14:
After having used this lib as intended quite a few times new, I would also like to use it in my new project, which makes use of a port expander (MCP23017). Originally I thought "Well, guess I have to fork OneButton for that", but looking into the code, it seems there could be an easier solution.
Add a constructor 
`OneBotton(int activeLow)` - that does not save a pin
and a method 
`void tick(int pinValue)` - which takes the new pin value (`void tick(void)` would then call `tick(digitalRead(_pin)))`

In the loop, you could then call all sorts of stuff like
`button.tick(digitalRead(_pin))` (same as button.tick())
`button.tick(digitalReadFast(_pin))` (e.g. for Teensy)
`button.tick(mcp.digitalRead(0))` (for the Adafruit MCP23017 lib)
...

Would you generally be interested in including something like this? I could make a pull request with the described changes.

schmersgahoven, 2017-10-12:
This would be great. I try this since days an struggle a lot be using OneButton with expander.
Kvoit is it possible that you provide me more information where you have input your additkions. I am not such firm in creating libraries and the depencies between the sketch himself and the libraries. So I alread tried to put the ne constructor OneBotton(int activeLow) and the method into the OneButton.h and the button.tick into the sketch. But my IDE do not like it. And I am not such experienced to find out why.
Thx for your help in advance

kvoit, 2017-10-12:
Haven't implmented this yet, actually. I have just forked this repo, since mathertel does not seem to be interested.  You can subscribe to my fork to be notified when I find time to do it.

kvoit, 2018-01-05:
I have implemented this now in my fork. It compiles, but I have not checked it on actual hardware. I will issue a pull request after I have, but it seems that this repository is more or less abandoned.

mathertel, 2018-09-26:
Implemented in version 1.3.0

closed 2018-09-26 by mathertel

Assarhadon96, 2021-01-20:
Im trying since weeks to make the Onebutton usable with an portexpander, is there an example or could somebody pls tell me how to add a button which is connected to a mcp23017 with the onebuttons funkctions ? i really like the onebutton lib but cant use it since i use port expanding. Thanks!

kvoit, 2021-01-21:
It looks like they implemented what I suggested above and did in the fork. So it pretty much says what to do ...

Assarhadon96, 2021-01-21:
> It looks like they implemented what I suggested above and did in the fork. So it pretty much says what to do ...

Yeah they are some changes, but how do I use it? The things u mentioned above are telling what to change in the library and how to change the tick function in void loop, but how I initialize the mcp pin? 

I tried it like this but then I got Problems with the watchdog reset

#define buttonexp 7 //pin on port expand

Onebutton Button (buttonexp, true, false);

Void setup()
{
Mcp.pinmode(buttonexp, INPUT);
}

Void loop()
{
Button.tick(mcp.digitalread(buttonexp)),
}