mathertel/OneButton #26: Debounce when "going up".

yunjova, 2017-12-18:
The debouncing is only doen when going to pressed state. There can also be bouncing when going up (e.g. some relay contacts can bounce in either way the switching goes).

So it would be good to add debouncing for the up states.

Kind regards,

S4mw1s3, 2017-12-19:
Isn't this solved here https://github.com/mathertel/OneButton/pull/20 ?

yunjova, 2017-12-21:
In state 3 there is no debounce for the release. There should be set a stopTime and then "wait" from that moment for an amount of debounceTicks.

In state 6 there is no debounce for the release.
  So a glitch on a long line is not covered. It will result in a "stop" of the long press.

I will need to look further how glitches (i.e. events not coming from a switch-press but from interference, ...) are handled as I have long lines in my application.

A side note: In state 6, in the else case, it is not needed to set _isLongPressed = true.

mathertel, 2021-01-24:
fixed in version 2.0.0

closed 2021-01-24 by mathertel