mathertel/OneButton #28: Single Click with Interrupt possible?

VolkerBo, 2018-02-06:
I tried to realize a solution with interrupt and **single click.**
But I can not get it to work.

My function works **every second single click** or **every double click**, but not on **single clicks**.

I want to realize a state machine that switches to the next state with every **single click**.

[TasterTest3.txt](https://github.com/mathertel/OneButton/files/1699355/TasterTest3.txt)

Can someone help?

Regards 
Volker

VolkerBo, 2018-02-07:
So I found out for myself.

[image]

One needs an edge to the transition to 1.
You need a second edge to transition to 2.
And finally you need a third flank to transition to click ().

Therefore you can not use "click ()" with interrupt.

Tell me if I am wrong.

Regards
Volker

gitmeister, 2018-03-30:
I'm having exactly the same behavior. I'm using the library PinChangeInterrupt (https://github.com/NicoHood/PinChangeInterrupt) and all works Ok except the click() is only called on the second button press. I don't see a reason other than tick() is only being called on transitions and the debounce code may want to see tick() with a consistent button state. I'll post if I get a solution.

gitmeister, 2018-03-30:
Here's a fix that i've tested works. Insert at Line 133 of OneButton.cpp:

```
  } else if (_state == 1) { // waiting for menu pin being released.

    if ((buttonLevel == _buttonReleased) && ((unsigned long)(now - _startTime) < _debounceTicks)) {
      // button was released to quickly so I assume some debouncing.
	    // go back to state 0 without calling a function.
      _state = 0;
// INSERT THIS >>>>     
    } else if ((buttonLevel == _buttonReleased) && (unsigned long)(now - _startTime) > _clickTicks) {
      // this was only a single short click
      if (_clickFunc) _clickFunc();
      _state = 0; // restart.  
// <<<< INSERT THIS
    } else if (buttonLevel == _buttonReleased) {
      _state = 2; // step to state 2
      _stopTime = now; // remember stopping time
```

foxsam21, 2020-01-31:
> Here's a fix that i've tested works. Insert at Line 133 of OneButton.cpp:
In the current code it would be line 206:
```
    if ((!activeLevel) &&
        ((unsigned long)(now - _startTime) < _debounceTicks)) {
      // button was released to quickly so I assume some debouncing.
      // go back to state 0 without calling a function.
      _state = 0;

  // INSERT THIS >>>> based on: https://github.com/mathertel/OneButton/issues/28#issuecomment-377613782   
    } else if ((!activeLevel) &&
        ((unsigned long)(now - _startTime) < _clickTicks)) {
      // this was only a single short click
      if (_clickFunc) _clickFunc();
      _state = 0; // restart.  
  // <<<< END INSERT THIS

    } else if (!activeLevel) {
      _state = 2; // step to state 2
      _stopTime = now; // remember stopping time
```

> PS: This is a quick hack since if you try to do a double click it won't work as expected as it'll give 2xclicks.

Is there anything anyone has done to get this working properly?

antno1000, 2020-03-03:
> 
> 
> > Here's a fix that i've tested works. Insert at Line 133 of OneButton.cpp:
> 
> In the current code it would be line 206:
> 
> ```
>     if ((!activeLevel) &&
>         ((unsigned long)(now - _startTime) < _debounceTicks)) {
>       // button was released to quickly so I assume some debouncing.
>       // go back to state 0 without calling a function.
>       _state = 0;
> 
>   // INSERT THIS >>>> based on: https://github.com/mathertel/OneButton/issues/28#issuecomment-377613782   
>     } else if ((!activeLevel) &&
>         ((unsigned long)(now - _startTime) < _clickTicks)) {
>       // this was only a single short click
>       if (_clickFunc) _clickFunc();
>       _state = 0; // restart.  
>   // <<<< END INSERT THIS
> 
>     } else if (!activeLevel) {
>       _state = 2; // step to state 2
>       _stopTime = now; // remember stopping time
> ```
> 
> > PS: This is a quick hack since if you try to do a double click it won't work as expected as it'll give 2xclicks.
> 
> Is there anything anyone has done to get this working properly?
> 
> Edit:
> It seems to be working fine for me without this edit... maybe it was fixed? (if so this should be closed...)

This code enables click but disables doublclick feature, can't use doubleclick now

mathertel, 2021-01-24:
Please check for version 2.0.0 and be sure to call tick() in the main loop() function, not only in interrupt routines.

closed 2021-01-24 by mathertel