mathertel/OneButton #43: where to find the description about instantiazation parameters int active, bool pullupActive ?

dsyleixa, 2018-11-28:
hello,
where to find the description about instantiazation parameters int active, bool pullupActive ?

 OneButton(int pin, int active, bool pullupActive = true);

IllyaMoskvin, 2019-09-10:
I agree that having documentation on these parameters would be good.

In general, though, it's best to get in the habit of reading library code.

Since the time this issue was opened, [comments were added](https://github.com/mathertel/OneButton/blob/f78433c20931eb0ec2df0a279692f06a4896fbf9/src/OneButton.cpp#L31-L52) that help explain what's going on:

```
OneButton::OneButton(int pin, int activeLow, bool pullupActive)
{
  // OneButton();
  _pin = pin;

  if (activeLow) {
    // the button connects the input pin to GND when pressed.
    _buttonPressed = LOW;

  } else {
    // the button connects the input pin to VCC when pressed.
    _buttonPressed = HIGH;
  } // if

  if (pullupActive) {
    // use the given pin as input and activate internal PULLUP resistor.
    pinMode(pin, INPUT_PULLUP);
  } else {
    // use the given pin as input
    pinMode(pin, INPUT);
  } // if
} // OneButton
```

mathertel, 2020-05-01:
See also the examples in the new Readme.md.

Thanks for contributing.

closed 2020-05-01 by mathertel