mathertel/OneButton #51: Avoid calling `pinMode` in constructor

IllyaMoskvin, 2019-09-10:
Currently, it seems that `pinMode` is [being set in the OneButton constructor](https://github.com/mathertel/OneButton/blob/f78433c20931eb0ec2df0a279692f06a4896fbf9/src/OneButton.cpp#L45-L51).

This seems to be bad practice:

- https://forum.arduino.cc/index.php?topic=384519.msg2650609#msg2650609
- https://forum.arduino.cc/index.php?topic=212844.msg1561234#msg1561234
- https://forum.arduino.cc/index.php?topic=261404.msg1845653#msg1845653
- https://forum.arduino.cc/index.php?topic=78114.msg590112#msg590112

(Ugh, sticky headers. Scroll up a bit after clicking these links.)

> The general advice is to have an init or begin function that is called from setup() after the object has been created.  This allows time for the hardware to settle down to a stable state before being used.

...more specifically:

> C++ constructors are called BEFORE main() (which makes sense, right?)
The AVR peripherals are initialized AFTER main()  (main() calls init(), from the core library.)
Therefore, constructors cannot do things to peripherals that are dependent on them already being initialized, or that will be un-done when the peripherals are initialized.  You should not do pinMode(pin, OUTPUT) in a constructor because other things needed before pinMode() will work might not have been done, and because init() might (sensibly) set the pins to the expected startup state (all inputs) afterwards...

> (now, in fact, init() on AVRs doesn't do anything with the gpio pins, counts on "all inputs" being left over from the HW reset, and pinMode() from a constructor would probably work just fine.  But you still shouldn't do it!  (especially because AVRs are not the only CPUs out there any more!)

I feel like this is dangerous if the `init()` causes your input to default to `INPUT_LOW` when you're expecting `INPUT_PULLUP` as defined in your `OneButton()` call, and then you connect it to ground. Aren't you risking creating a short circuit?

It seems that other libraries have solved this by creating a separate method from the constructor, e.g. `begin()` that's meant to be called inside `setup()`. The constructor just creates an empty object. It's the begin method that would accept `pin`, `activeLow`, etc.

I'm new to this, so I'm not sure. Thoughts?

mathertel, 2019-10-21:
Thanks for the hints and your thoughts.

I never had a problem with initializing the pin mode during construction but I will study the links you added here.

INPUT_PULLUP is not creating a shortcut because it is adding an 20k resistor to set the input level on HIGH when no other signal is present at the pin. See: https://www.arduino.cc/en/Tutorial/DigitalPins

JAY-M-ERICSSON, 2020-01-14:
This is indeed a potential challenge as constructor are called before the auto-generated main() function where Arduino IDE calls init() and initVariant(). Those functions could undo your pin setup but luckily they don't at this stage (they do set up the timers etc) hence why you don't see any issue. 

hence the best practice to have a begin() function that you call in setup(), at that stage you are sure that main() has been called and what you set is under your control.

it would not be difficult to modify your code and add such a begin() method. In order to maintain compatibility with the past, your constructor would call begin() and the begin() function would have your existing constructor code (minus instance variables assignment)

```
void OneButton::begin(bool pullupActive = true)
{
  if (pullupActive) {
    // use the given pin as input and activate internal PULLUP resistor.
    pinMode(_pin, INPUT_PULLUP);
  } else {
    // use the given pin as input
    pinMode(_pin, INPUT);
  } // if
}

OneButton::OneButton(int pin, int activeLow, bool pullupActive)
{
  // OneButton();
  _pin = pin;

  if (activeLow) {
    // the button connects the input pin to GND when pressed.
    _buttonPressed = LOW;

  } else {
    // the button connects the input pin to VCC when pressed.
    _buttonPressed = HIGH;
  } // if

  // forward compatibility 
  begin(pullupActive);
} // OneButton
```

this way existing code that does not call begin() would still work as in the past, and new code could add a begin(pullupActive) call. To make it just a begin() would require adding an instance variable to memorize de pull-up mode, which would cost 1 extra byte. 

You could save that byte with a bit of work and reusing your existing _buttonPressed instance a bit differently. Today you declare this as an int, which is 2 or 4 bytes and is a waste of space as you only need to store 1 bit, HIGH or LOW. 

So you could change _buttonPressed wherever you read it today into (_buttonPressed &0b1)  and use the first bit (LSB) as your HIGH or LOW indicator,  and only declare it as a uint8_t 

Then you have 7 more bits to play with to store other information 

just an idea :)

mathertel, 2021-01-24:
you can also create the OneButton instance in the setup() if you like.
Just use
```CPP
OneButton *button;

button = new OneButton(...);
```
and then use an arrow not the dot notation.
button->attach...(...);

It is used in the example SpecialInput in version 2.

closed 2021-01-24 by mathertel

JAY-M-ERICSSON, 2021-01-25:
that's one way to do so. I find it sub-optimal but that's your (great) library 👍