mathertel/OneButton #52: Bug on the website example code

Gilligan8, 2019-10-20:
It says:

OneButton button(A1);

I spent a LOT of time trying to understand libraries, using yours as an example, and couldn't understand how that was supposed to work.

Then I tried compiling it and found that it wouldn't compile because that line is missing an argument.

Just thought I'd pass the info along.

mathertel, 2019-10-21:
Please use the latest version of the library.

This line creates a button instance that uses the A1 as input line.

closed 2019-10-21 by mathertel

Gilligan8, 2019-10-21:
I understand that... but people will still come across that website as it's referred to as the place to go to understand the library.

Seems like just adding that ", true" part would be pretty simple and solve that confusion. :/

mathertel, 2020-05-01:
Starting with 1.5.0 the initialization has another default parameter.
The readme.md was updated to have more examples.