mathertel/OneButton #55: Debounce tick times

Rolbista, 2020-01-13:
Hello

As far as I can see there is no way to set:
```
unsigned int _debounceTicks
unsigned int _clickTicks
unsigned int _pressTicks
```
whether via constructor or later during runtime, can you confirm this? If so, would you consider adding such a functionality or accepting a PR that adds it?

JAY-M-ERICSSON, 2020-01-14:
I see [those functions in the API](https://github.com/mathertel/OneButton/blob/c8651328e3fd3513ffac95f5e012ebaf9a2b51e5/src/OneButton.h#L46-L53):

`  // set # millisec after safe click is assumed.
  void setDebounceTicks(int ticks);

  // set # millisec after single click is assumed.
  void setClickTicks(int ticks);

  // set # millisec after press is assumed.
  void setPressTicks(int ticks);
`

isn't that what you are looking for?

Rolbista, 2020-01-14:
Yes, my bad. Tested and works fine, thank you.

closed 2020-01-14 by Rolbista