mathertel/OneButton #61: longPress doesn't invert for active HIGH

Craigzyc, 2020-05-16:
Hi,

I am using this library to monitor one of my buttons to use the longPress detection.

The button is active HIGH and setup using 
`OneButton selectButton(17, HIGH, false);
`
It worked fine on my breadBoard using active LOW, but on the board I made its active HIGH since theres some other circuitry. 
When I changed it to active HIGH the attachClick is now called correctly but the longPress still seems to be active LOW.
Is anyone else experiencing this? ESP32

underwoodblog, 2020-05-27:
Hi,

I've problems using active LOW like in the docu for button to GND with internal pullup.

```
OneButton btn = OneButton(
  BUTTON_PIN, // Input pin for the button
  LOW,       // Button is active LOW
  true       // Enable internal pull-up resistor
);
```

With LOW, longpressstart and stop fires with every button click.

Craigzyc, 2020-05-27:
@underwoodblog 
I've seen that. Likely means your trying to call it on a pin that doesn't actually have an internal pullup.
GPIO 34-39 do not have this feature. Are you trying to use one of them?

mathertel, 2020-05-27:
(I have to add more doku)
In the OneButton.cpp file you can find the right description to the 2. parameter.
Don not pass HIGH and LOW but true and false:

// The 2. parameter activeLOW is true when the external wiring sets the button to LOW when pressed.
OneButton button(PIN_INPUT, true);

// The 2. parameter activeLOW is false when the external wiring sets the button to HIGH when pressed.
// The 3. parameter can be used to disable the PullUp .
OneButton button(PIN_INPUT, false, false);

closed 2020-05-27 by mathertel