mathertel/OneButton #63: BUG: wrong time of double click event raising

ppetrov-dev, 2020-05-24:
```
#include <Arduino.h>
#include "OneButton.h"
OneButton _button = OneButton(4, true, true);
void setup()
{
    Serial.begin(9600);
    Serial.println("Serial OK");
    _button.attachDoubleClick([]() { Serial.println("Double Click"); });
}

void loop()
{
    _button.tick();
}
```
**STR**:
1. double click (don't release button after second click)
2. wait for a while (no matter, 1 sec, 5 sec or 1 hour)
3. release the button.

**Actual result**: "Double Click" is in output after releasing the button
**Expected result**: "Double Click" is in output after second click detected

ppetrov-dev, 2020-05-24:
Maybe after 1st step the object should raise a double click event then raise a long-press start event. Or introduce a new event "double press start) or something=) What do you think?

mathertel, 2020-05-27:
The double click is detected when the button is released. That's by design and the common approach.
See https://www.w3schools.com/jsref/event_ondblclick.asp as a cross reference to doubleclick in a browser button.

closed 2020-05-27 by mathertel