mathertel/OneButton #69: Minimum value of setClickTicks for double click feature

hoilett, 2020-07-19:
It might be useful to note somewhere in the code that it will be hard to detect a double click unless setClickTicks is at least 200 milliseconds (or so). Anything shorter than 200ms and there just isn't enough time for the user to press a button twice. I initially had setClickTicks to 10 ms in order to make the button very responsive. When I tried adding a double click feature, I noticed the double click callback function simply would not register, even when I tried to click as fast as possible. I had to increase setClickTicks to at least 200 and then I eventually set it to 275 ms which I thought was practically more comfortable.

It makes sense that setClickTicks needs to be relatively long, but it wasn't immediately obvious. Just a suggestion. Really great library.

mathertel, 2021-01-24:
Thanks, I added some comments to the readme.
in version 2 the default will be 400/800.

closed 2021-01-24 by mathertel

hoilett, 2021-01-24:
@mathertel awesome. Thanks.