mathertel/OneButton #75: Passing a GPIO when using an MCP23s17

Mesihas, 2020-12-13:
Don't have free GPIO and  I am using a 23S17 which gives me extra 16 digital GPIO but don't know how to pass the parameter.

`OneButton button(A1, true); ` 

my GPIO is:
mcp23s17.pinMode( button8, INPUT_PULLUP );

ricreis394, 2020-12-28:
I don't think that's possible in this library

kvoit, 2021-01-21:
Have you checked #24  ?

Assarhadon96, 2021-01-23:
got somebody a solution for that? i dont understand how its working

mathertel, 2021-01-24:
It is possible to use the library without the built-in (and limited to ) direct digital input.

There is a new example "SpecialInput" in version 2.0 to demonstrate:

```CPP
#include "OneButton.h"

// This is an example on how to use the OneClick library on other input sources than standard digital pins.
// 1. do not use a pin in the initialization of the OneClick library.
// 2. pass the input state to the tick function.

#if defined(ARDUINO_AVR_UNO)
#define PIN_INPUT 2

#else if defined(ESP8266)
#define PIN_INPUT D3

#endif

// create a OneButton instance without pin.
OneButton *button = new OneButton();

void setup() {
  Serial.begin(115200);
  Serial.println("One Button Example with custom input.");

  // Here is an example on how to use an inline function as parameter:
  button->attachClick([]() { Serial.println("Click"); });
  button->attachDoubleClick([]() { Serial.println("DoubleClick"); });

  // setup your own source of input:
  pinMode(PIN_INPUT, INPUT_PULLUP);

} // setup()

void loop() {
  // read your own source of input:
  bool isPressed = (digitalRead(PIN_INPUT) == LOW);

  // call tick frequently with current push-state of the input 
  button->tick(isPressed);
} // loop()
```

closed 2021-01-24 by mathertel

Mesihas, 2021-11-13:
Thanks for this,  it works very well.