mathertel/OneButton #77: question with interrupts

floydheld, 2021-01-29:
Hi!
First: I like your code! Seems really neat! Just earlier today I decided to look for a library with the exact same features and here it is, handcrafted only days ago... wow!
No my question is, as I would like to extend on your "interrupt" example, ...
Oh.. while writing this I realized my queston doesn't make sense...
Just a note: If you use an interrupt to call tick() you also can't use Serial in the callbacks singleClick, doubleClick, ... Because the callbacks are also run from the interrupt.
Thanks again ;)

mathertel, 2021-01-30:
I had this idea as well and mentioned it in the CHANGELOG.md.
From the state diagram you can see that the callback functions are always triggered by a timing condition and not by a input change (reported through interrupts).
So maybe this is not a problem - but let's do a test....

mathertel, 2022-01-21:
no more input into this finding.

closed 2022-01-21 by mathertel