mathertel/OneButton #78: Compile Warning

PowerBroker2, 2021-01-30:
My sketch has been working great before this last update. I'm getting the following compile warning:

```
\Arduino\libraries\OneButton\src\OneButton.cpp: In member function 'void OneButton::tick(bool)':
\Arduino\libraries\OneButton\src\OneButton.cpp:223:10: warning: enumeration value 'UNKNOWN' not handled in switch [-Wswitch]
   switch (_state) {
          ^
Sketch uses 35712 bytes (1%) of program storage space. Maximum is 2031616 bytes.
Global variables use 49844 bytes (9%) of dynamic memory, leaving 474444 bytes for local variables. Maximum is 524288 bytes.
```

mathertel, 2021-01-30:
What board /  compiler are you using ?

PowerBroker2, 2021-01-30:
Sorry, Teensy 4.0 with the latest Arduino IDE version on Windows 10

PowerBroker2, 2021-01-30:
Ok, strangely enough the warning has gone away without editing the lib files... interesting to say the least

closed 2021-01-30 by PowerBroker2

mathertel, 2021-01-31:
The warning appears only when compiling with the option to show ALL warnings.
I'll  take care.