mathertel/OneButton #88: Setting the attachDuringLongPress event interval

FransCouwberghs, 2021-04-16:
Is it possible to set the interval for the attachDuringLongPress event?
I use two buttons to increase or decrease a number on a display.
I use the same function for attachClick and attachDuringLongPress so one can increase/decrease with one digit or 'speed' true the setting by holding down the button.
The 'speed' however is way to fast.
What I do now is testing at the end of the function whether the button is in a isLongPressed()) state and delay a period:

if (btnIncrease.isLongPressed()){
    delay(50);}

This works quite well, but I have a aversion for delay().
I was thinking of a function setLongPressInterval(int) (likewise setPressTicks(int)) to set the interval.

Keep up the good work!

Zaighum1, 2022-12-05:
have you find it??? if yes then kindly guide me if i can set the interval and duration of attachDuringLongPress event ??

IhorNehrutsa, 2023-05-11:
See https://github.com/mathertel/OneButton/examples/LongPressEvents/LongPressEvents.ino

mathertel, 2023-06-03:
fixed by https://github.com/mathertel/OneButton/pull/124

closed 2023-06-03 by mathertel