mathertel/OneButton #89: Interrupt not reliable

SamuelReinfelder, 2021-04-16:
I tried the interrupt example with a simple single click button.

It does work, but it only works every second time or so. If I put 

`btn_encoder.tick();`

in the loop function, it works every time. I am guessing it has something to do with the debouncing, but I am not sure.

SamuelReinfelder, 2021-04-16:
Ah I think I get it. The state machine implemented in OneButton.cpp cant really work with Interrupts.

It needs to be called several times to advance the states. It is also important when to call it. The interrupt calls it at least twice, but this may not be enough or not at the right time.

Does it work for you?

mathertel, 2022-01-21:
As in the example and by your finding the button.tick() MUST be called in loop as well.

closed 2022-01-21 by mathertel

cyberman54, 2022-11-06:
Thus, what purpose has the given interrupt example? Is there any advantage to trigger the state machine additionally by interrupt, maybe a bit more responsiveness?