mathertel/OneButton #94: doubleclicking results in singleclick

linxcow, 2021-09-08:
My finger doubleclick speed is probably too fast for the default values , i have to slow my doubleclicking to make it work.
Do others have the same problem , might be interesting to change the default values in the library , to broaden the timing range.

linxcow, 2021-09-08:
Changing the debounce time back to 50ms from 80 in the example made it a lot better

mathertel, 2021-09-08:
The debounce time is required to compensate high frequent HIGH-LOW-HIGH-LOW-... sequences when a momentary button
gets opened or closed. The is caused by "bad" hardware and the duration depends on physical measurements like elasticity of the contact material and bad connections caused by dirt.
There is no "Best" timing but you can adjust - as you did - to your specific hardware.
Adding a small capacitor in parallel to the button also helps.

mathertel, 2022-01-21:
I just saw a dev board with the following components:
* 10k resistor from VDD3v3 to button-side 1
* 0.1µF capacitor from button-side 1 to GND
* button-side2 to GND.  
Seems to be a reasonable choice for debouncing manual buttons.

closed 2022-01-21 by mathertel