mathertel/OneButton #103: ESP32 and FreeRTOS

mitchsf, 2021-12-30:
Has anyone else tried this? When tick() is run in a task, the function crashes with errors that I'm not clear on. Lines 56 and 199 of OneButton.cpp indicate it has something to do with pinMode or digitalRead, according to the Exception Decoder.

It works perfectly if tick() is in loop(), but with FreeRTOS, it's not the best idea to have anything running there.

mitchsf, 2021-12-30:
I tried using a semaphore to block anything else from running, to no effect.

`void RunFunctions( void * parameter) {
  for (;;) {
    xSemaphoreTake(buttonMutex, portMAX_DELAY);
    button.tick();
    xSemaphoreGive(buttonMutex);
    vTaskDelay(10);
  }
}`

mathertel, 2022-01-15:
I am not aware of such a problem.
Can you please investigate more and share your findings ?

mitchsf, 2022-01-17:
Sorry, I don't know enough about FreeRTOS to figure this one out. I just left tick() in loop().

closed 2022-01-21 by mathertel