mathertel/OneButton #110: how to use this library with MCP23017??

Alexandre2003, 2022-11-16:
how to use this library with MCP23017 has how ??

abishur, 2023-04-21:
Did you ever figure this out?

Alexandre2003, 2023-04-24:
Hello, so I wanted to use the OneButton library with the mcp23017 pins, as I didn't find anything about it in the topics, I ended up giving up

mathertel, 2023-04-25:
The library is not made for using input signals from other sources but the DigitalIn Pins.
If you like to use the MCP23017 as input you have to re-write parts of the library.
It can be done like I did it in https://github.com/HomeDing/HomeDing/blob/develop/src/ButtonElement.cpp.
The loop() function in this implementation is doing almost the same as the button library but is not using the digital read directly.

As a general comment, this library can be used under conditions and assumptions and this is true to many other libraries. Coding skills are required.
Feel free to just use part of the library code for your specific problem.

Regarding the Homeding library and eco-system:
An Element for the  mcp23017 can be implemented to create actions based on input levels and send them to the button element. This is a action based implementation approach that fits well to IoT Implementations.

closed 2023-04-25 by mathertel

ChristianKnorr, 2023-05-01:
Yes. Look:
`
/*
 * MCP23017Encoder_OneButton.ino - Example for the OneButtonLibrary library.
 * This is a sample sketch to show how to use the OneClick library on MCP23017 pins.
 *
 * The library internals are explained at
 * http://www.mathertel.de/Arduino/OneButtonLibrary.aspx
 *
 * Setup a test circuit:
 * * Connect 2 pushbuttons to pin 2 at mcp0 and ground, and pin 5 at mcp1 and ground.
 *   
 * The sketch shows how to setup the library and bind the functions (singleClick, doubleClick) to the events.
 * In the loop function the button.tick function must be called as often as you like.
 *
 * * 22.01.2021 created by Matthias Hertel
*/

#include "OneButton.h"
#include <Adafruit_MCP23X17.h>
Adafruit_MCP23X17 mcp0;
Adafruit_MCP23X17 mcp1;

// This is an example on how to use the OneClick library on other input sources than standard digital pins.
// 1. do not use a pin in the initialization of the OneClick library.
// 2. pass the input state to the tick function.

// You can also find how to create an instance in setup and not by declaration.
// You can also find how to use inline callback functions.

// OneButton instance will be created in setup.
OneButton *button1;
OneButton *button2;

void setup() {
  mcp0.begin_I2C(0x20); // i2c address 0x20
  mcp1.begin_I2C(0x21); // i2c address 0x21
  mcp0.pinMode(2, INPUT_PULLUP);
  mcp1.pinMode(5, INPUT_PULLUP);
  Serial.begin(115200);
  delay(1000);
  Serial.println("One Button Example with custom input.");

// create the OneButton instance without a pin.
  button1 = new OneButton();
  button2 = new OneButton();

  // Here is an example on how to use a parameter to the registered function:
  button1->attachClick([]() { click(1); });
  button1->attachDoubleClick([]() { doubleclick(1); });
  button1->attachMultiClick([]() { multiClick(1, button1->getNumberClicks()); });
  button1->attachLongPressStart([]() { longPress(1); });
  
  button2->attachClick([]() { click(2); });
  button2->attachDoubleClick([]() { doubleclick(2); });
  button2->attachMultiClick([]() { multiClick(2, button2->getNumberClicks()); });
  button2->attachLongPressStart([]() { longPress(2); });

} // setup()

void loop() {
  // read your own source of input:
  bool isPressed1 = (mcp0.digitalRead(2) == LOW);
  bool isPressed2 = (mcp1.digitalRead(5) == LOW);

  // call tick frequently with current push-state of the input
  button1->tick(isPressed1);
  button2->tick(isPressed2);
} // loop()

void click(int b) { // Parameter to identfy wich button pressed
  Serial.print("Button");
  Serial.print(b);
  Serial.println(" click");
} // click
void doubleclick(int b) { // Parameter to identfy wich button pressed
  Serial.print("Button");
  Serial.print(b);
  Serial.println(" doubleclick");
} // doubleclick

// this function will be called when the button was pressed multiple times in a short timeframe.
void multiClick(int b, int n) { // Parameter to identfy wich button pressed and how often
  Serial.print("Button");
  Serial.print(b);
  Serial.print(": ");
  if (n == 3) {
    Serial.println("tripleClick detected.");
  } else if (n == 4) {
    Serial.println("quadrupleClick detected.");
  } else {
    Serial.print("multiClick(");
    Serial.print(n);
    Serial.println(") detected.");
  }
} // multiClick

void longPress(int b) { // Parameter to identfy wich button pressed
  Serial.print("Button");
  Serial.print(b);
  Serial.println(" long press");
} // longPress
`

mathertel, 2023-05-07:
Thanks for the code.
Yes, seems correct working.
Can you create a pull request to add the example ?