mathertel/OneButton #113: possible to add quick press + long press?

madsi1m, 2022-12-25:
Hi is it possible to add call back for a quick press followed by a long press as an event?

For example, like click and drag on a touchpad.

jazztunes, 2024-04-04:
FYI, The "quick press followed by a long press" combo is what DJI uses to power toggle their drones. I think it's safer because it requires a bit more intentionality to get it switch on or off.

mathertel, 2024-06-02:
There is a new "press" event added to the library in https://github.com/mathertel/OneButton/pull/141.
This will be part of the next release.

closed 2024-06-02 by mathertel

jazztunes, 2024-06-05:
Thanks so much for looking at this issue!

When you do the docs for the new release, can please you add an example of how to use the new "press" event to accomplish "quick press followed by a long press"?  Thanks again!