mathertel/OneButton #125: Update Published PlatformIO version

sachaw, 2023-05-14:
Currently the registry has 2.0.3 as the latest, can this please be updated?

caveman99, 2023-05-15:
PIO should have picked that up by now, relevant change was 3 days ago - maybe the JSON Change needs to be included with the 2.1.0 tag as well as Master?

vshymanskyy, 2023-08-09:
@mathertel may I kindly ask you to do a library release for both Arduino and PlatformIO?
Specificly, it would be useful to have changes like this: https://github.com/mathertel/OneButton/commit/5e7791bafaa371e71cfc0862f11b6655d3ecd184
Also it would tackle questions like #127 

Best wishes!

the-skips, 2023-11-11:
2.0.3 is still the latest version on PlatformIO unfortunately. It is pretty popular (790 downloads last month) so an update would be appreciated.

ivankravets, 2023-11-27:
The [2.1.0](https://github.com/mathertel/OneButton/tree/2.1.0) tag contains manifests with two different versions. The `library.json` points to `2.0.3` where `library.properties` to the `2.1.0`.

PlatformIO can't handle such types of conflicts in automatic mode. Once @mathertel updates both manifests to the new version and makes a new tag, PlatformIO will automatically pick the new version and update it in the registry.

mathertel, 2023-12-02:
Version 2.5.0 should fix this and platform.io version is in sync.

closed 2023-12-02 by mathertel