mathertel/OneButton #128: Lag after button press

MickeyGentry, 2023-07-12:
This library worked as expected immediately, but there is a lag after pressing a button. Maybe half a second. This was with the example code. Any thoughts as to why?

mathertel, 2023-07-13:
This is by design, because:

If you register a doubleClick or multiClick function the library waits after the first click
for a possible second on n-th click following.

E.g. it is not intended in case of a doubleClick input pattern to report a singleClick event first.

Please check if the delay is there when not registering on doubleClick or multiClick events.

MickeyGentry, 2023-07-13:
Thanks for getting back, I will check that out.

My usage is for a simple MIDI controller pedal; two momentary foot switches
and a rotary. I’m trying a few libraries and some are just overkill and too
large for my needs. Not yours!

Each switch must do single, double and long press. Pushing both buttons
together will be a 4th event. I’m pretty sure your library can do all of
that but the ‘lag’ due to my misunderstanding blocked me.

I’ll let you know how it goes.

Michael

MickeyGentry, 2023-07-13:
Confirmed, using  `button1.attachClick(click1);` by itself shows now lag. Unfortunately I need all the functions on one button so the lag will not work for a music related program change pedal.

thanks!

MickeyGentry, 2023-07-13:
Welp I’m back . I may forgo multi clicks because everything else works well. This is supposed to be a foot switch so not using double clicking is probably a good thing! I’ll let you know how it goes.

thanks!