mathertel/OneButton #143: How to get rid of #define PIN_INPUT?

mister-Monk, 2024-06-27:
I want to use the GPIO number obtained from SPIFFS in OneButton.
Something like this:

#include <OneButton.h>
#include <FS.h>

void setup()
{
	File _file = SPIFFS.open
	...
	BUTTON_pin = r_c.toInt();
	...
	OneButton button1(BUTTON_pin, true, true);
}

Is this possible? How?

mister-Monk, 2024-06-27:
This is what I did:

Added a copy OneButton::OneButton to the library

  void initOB(const int pin, const boolean activeLow = true, const bool pullupActive = true);
  
  void OneButton::initOB(const int pin, const boolean activeLow, const bool pullupActive)
{
  // OneButton();
  _pin = pin;

  if (activeLow) {
    // the button connects the input pin to GND when pressed.
    _buttonPressed = LOW;
	...

Further

#include <OneButton.h>
#include <FS.h>

void setup()
{
	File _file = SPIFFS.open
	...
	BUTTON_pin = r_c.toInt();
	...
	button1.initOB(BUTTON_pin, true, true);
}

It works. But I'm not sure if this is the right way.

mathertel, 2024-07-31:
There is no need to use a OneButton instance as a global variable. You can defer creating this by just having a global pointer and
initializing the OneButton instance later.

See current version of the example SimpleOneButton.

This change was required because of a changed behavior in the Arduino library of Arduino Nano ESP32
where initialization also had to be delayed. See Issue #129

closed 2024-07-31 by mathertel